antonblanchard/microwatt #1: add instructions for FPGA implementation

sharkcz, 2019-08-23:
First, thanks for doing this all. Second, I have verified that your core really returns 2 for a "1+1" command issued in micropython "natively" on a Power9 Talos II running Fedora 30 for ppc64le as host :-) But it would be great if you could publish also similar instructions for people wanting to play with the FPGA implementation.

antonblanchard, 2019-08-23:
Thanks for trying it out! I agree we need to add some instructions on how to get it onto an FPGA.

Right now we load all of micropython into block ram, and that sets a somewhat unreasonable low bar for the FPGA (1MB of BRAM). We should be able to cut micropython down some more which will help. Do you have an FPGA to try out? So far I have used Xilinx FPGAs so I can put together instructions on how to do it with Vivado.

sharkcz, 2019-08-23:
I have 2 rather old Xilinx Spartan3 boards and I'm planning to buy the Nexys Video board you have mentioned in the keynote. So no problem for me with focusing on Xilinx devices, except there is no Vivado for ppc64le (yet) :-) And the open tools (Yosys) are still WIP.

olofk, 2019-08-23:
I added support for building for Nexys A7 and Nexys video with FuseSoC in https://github.com/antonblanchard/microwatt/pull/2

antonblanchard, 2019-08-23:
The Nexys Video is a pretty expensive board, although it does have a lot of stuff on it. When I get back home we are going to work on cutting things down so it works on a much broader (and cheaper) range of FPGAs. The work from @olofk will also help make it more broadly usable.

A few people are looking at getting this into verilog so we can start using the open tools too.

1138-4EB, 2019-08-24:
> A few people are looking at getting this into verilog so we can start using the open tools too.

An alternative to rewritting it in verilog might be [tgingold/ghdlsynth-beta](https://github.com/tgingold/ghdlsynth-beta), which allows to use ghdl as a frontend for yosys. There is also `ghdl --synth`, but the output is not usable yet. In any case, ghdl needs to be built with the following configure options: `--enable-libghdl --enable-synth`.

ghdlsynth-beta, and synthesis support in GHDL, is early experimental, so I'd expect this project not to be synthesisable now. But having it as a test can be very valuable.

sharkcz, 2019-08-27:
Sounds great, thanks.
FWIW I have `fusesoc` and its deps packaged for Fedora. After polishing it up I will publish it, likely in my copr repository and try to include it in Fedora directly.

olofk, 2019-08-27:
Sorry for going of topic, but @sharkcz, feel free to contact me if you have any questions regarding the FuseSoC packaging. Happy to help out

@antonblanchard I would recommend going with the LiteDRAM DDR controller. Using it for the SweRVolf SoC on Nexys A7 and can help getting it running on other devices. You would also need an SPI controller or a debugger to load larger programs from Flash. Not sure what the debug infrastructure looks like on power, but feel free to reach out if you want to throw around ideas. I have done this on both OpenRISC and RISC-V previously

antonblanchard, 2019-08-27:
@sharkcz There are initial instructions in the README now. Hopefully it will also run on lower end FPGA boards with the A7-100T (eg the bigger Arty A7 board). That is still to be confirmed however.

We still plan to put micropython on a diet, and get some form of DRAM controller integrated. We definitely want to get this onto a sub $100 device.

antonblanchard, 2019-08-27:
@olofk I was just about to ask you about this :) It's about time we tried to get the DRAM going on the Digilent boards. Could we also pull in a more standard UART?

Should I start by looking at SweRVolf?

ozbenh, 2019-08-28:
For loading to dram I was going to put together a little bootp/tftp stack on top of liteeth when I find a bit of time, but loading from an sdcard in spi mode should be doable too.

As for debug, no real standard worth looking at in powepc land, you don't want riscwatch :-) I was going to look at what riscv does. I had a USB tiny-h somewhere if I can find it...

I have the core using about half of an Arty (original one with A35T) not counting block ram. Without DDR I can get 128KB of RAM in blockram with a bit to spare, but I'd rather keep it for caches & tlb :-)

I was going to toy with a mig generated one with an axi<->wb bridge to get started but litedram sounds better.

ozbenh, 2019-08-28:
Allright, I google around a bit ... ;-) We should probably hookup a simple debug "bus" onto a jtag chain which can then either be hooked up on opencore's TAP or whatever TAP one wants, or BSCAN2 on Xilinx so we don't need extra HW on Arty.

antonblanchard, 2019-08-29:
@1138-4EB I took a look at ghdlsynth-beta (using the docker image), but I got a bit lost when ghdl-mcode failed to produce any output. I submitted https://github.com/ghdl/ghdl/issues/903 to track it although there's a good chance I'm doing something stupid.

1138-4EB, 2019-08-29:
I replied in ghdl/ghdl#903. As said, it is ok for ghdl-mcode to not produce any output/binary when running `ghdl -e`. However, for synthesis you don't need to use either `ghdl -a` or `ghdl -e`. In order to synthesise with the yosys plugin, just execute `yosys -m ghdl -p 'ghdl afile.vhdl anotherfile.vhdl lastfile.vhdl -e top_entity; synth_ice40 -json output.json'`.

The 'regular' GHDL binary is provided in `ghdl/synth:beta` for two reasons:

- A working GHDL installation is a dependency of the ghdlsynth yosys module. The ghdl binary itself is not executed by yosys, but many other pre-built libraries are used.
- Some users might want to first simulate and then synthesize the design, as you apparently did.

For completeness, `ghdl --synth` is available in both images, `ghdl/synth:latest` and `ghdl/synth:beta`. Nevertheless, the output of `ghdl --synth` is not usable yet.

carlosedp, 2019-09-15:
Hi guys, How big would be the effort to run microwatt on a different fpga? I have a Microsemi Polarfire and would love to try this on it. Thanks.

antonblanchard, 2019-09-20:
Hi @carlosedp, I don't have much experience with the Microsemi parts. I'm not sure if fusesoc supports the Microsemi toolchain, but if not you could manually import the project. We try to keep the VHDL portable but we have had to fix a few things along the way when trying out new compilers.

I will note that it is pretty unoptimised at the moment, so the resource consumption is a bit embarrassing - something we plan to fix. As such you might struggle to get it to fit on a smaller FPGA.

carlosedp, 2019-09-20:
The one I have is pretty beefy, like 300k LUTS and all that jazz :)
https://www.microsemi.com/existing-parts/parts/138273

I'll fire up it's software and try to load the VHDL for microwatt. I might need some help on how to port in case it's needed :)

Thanks!

sharkcz, 2019-10-03:
Since a while ago, fusesoc is available for Fedora from https://copr.fedorainfracloud.org/coprs/sharkcz/danny/, feedback is welcome.

sharkcz, 2020-01-17:
I think we can close this one. I have verified that the procedure from README works, even managed to program the flash chip on the Nexus Video board with the Micropython image :-)

closed 2020-06-11 by mikey