antonblanchard/microwatt #8: Use a proper open source software license

freemin7, 2019-08-28:
Creative commons is not meant for licensing software. 

https://creativecommons.org/faq/#can-i-apply-a-creative-commons-license-to-software

It is even in doubt if it is an open source license, since it is not listed here:
https://opensource.org/licenses/alphabetical

1138-4EB, 2019-08-29:
> Creative commons is not meant for licensing software.
>
> https://creativecommons.org/faq/#can-i-apply-a-creative-commons-license-to-software

It is arguable whether a project with >90% of the sources written in VHDL should be defined as a "software project". This issue is still legit, because CC licenses are not meant for hardware neither. This is mentioned in https://creativecommons.org/faq/#what-are-creative-commons-licenses:

> CC licenses may be applied to any type of work, including educational resources, music, photographs, databases, government and public sector information, and many other types of material. **The only categories of works for which CC does not recommend its licenses are computer software and hardware**.

---

> It is even in doubt if it is an open source license, since it is not listed here:
> https://opensource.org/licenses/alphabetical
> https://www.gnu.org/licenses/license-list.html

Creative Commons is a group of licenses, so you will not find them as a set. You need to search for the specific CC variant that is used. In this case, CC-BY. It is listed (green) in the gnu site you linked: https://www.gnu.org/licenses/license-list.html#ccby`. It says:

> This is a non-copyleft free license that is good for art and entertainment works, and educational works. It is compatible with all versions of the GNU GPL; however, like all CC licenses, it should not be used on software.

Note that 'non-copyleft' refers to the fact that it does not include the 'sharealike' clause. Still, it is a free (as in free speech) and open source license. See http://freedomdefined.org/Licenses#Comparison_of_Licenses

Regarding opensource.org, although creative commons licenses are not listed (I wonder why), all the website is licensed under CC-BY.

---

Unfortunately, the list of available free and open hardware licenses for hardware description projects is almost non-existent: https://en.wikipedia.org/wiki/Open-source_hardware#Licenses. As you see, most of the HDL projects are doomed to use either 'software licenses' or 'artistic licenses'. It is no better/worse to use GPL or CC-BY-SA. None of them fit.

There is this CERN Open Hardware License (OHL) (https://www.ohwr.org/project/cernohl/wikis/home) and they are looking at fixing it in v2 to make it fit better for HDL projects: https://wiki.f-si.org/index.php/CERN_OHL_v2_draft.

There is also the TAPR license. See https://www.tapr.org/ohl.html and https://en.wikipedia.org/wiki/TAPR_Open_Hardware_License. *"Like all Open Hardware Licenses, the OSI has not chosen to review the TAPR license"*.

---

Overall, unless you mean explicitly license the ~6% of C and Python sources in this repo, I think that, currently, CC-BY is as good as any other alternative.

freemin7, 2019-08-29:
The CC-BY also requires attribution. One could also explore the notion about removing that requirement.
Why? I recall a talk given by Limor Fried about open source hardware and problems they faced.
The x0xb0x was an open hardware re-implementation of a Roland synthesizer. The design was adapted and improved by several companies. After some time there were some clones were out there which violated the attribution requirement and when contacting them they could not be convinced to put the attribution on there. https://youtu.be/UYRhupdnUcY?t=696 
Why now? It is easier to change the license when the project is young.

1138-4EB, 2019-08-29:
I think that attribution is inherent to licensing. All the copyright, copyleft, public domain... environment is based on attribution. See https://opensource.stackexchange.com/questions/4577/does-gplv3-require-attribution

The case you reference about x0xb0x is how copyright works. If you claim to be the owner of the copyright of a product, you need to prove it. If you don't, others can just ignore your claim. There are frequent news about companies going to court because of this kind of issues.

From this point of view, it is not possible to remove the requirement of attribution. However, it is possible that the specific details about how CC-BY requires it to be done might not be suitable for this project. If this is the case, I'd please ask you to propose some specific alternative(s) which does/do have more suitable procedures.

> Why now? It is easier to change the license when the project is young.

I agree on this.

> Same reservations exists about non commercial licenses from her: https://youtu.be/Ca1dbM582Sc?t=222

Non-commercial clauses make licenses non-copyleft and non-free:

> Freedom 0: The freedom to run the program for any purpose.

freemin7, 2019-08-29:
> I think that attribution is inherent to licensing. All the copyright, copyleft, public domain... environment is based on attribution.
I think this is wrong since there are several licenses that work without attribution:
https://en.wikipedia.org/wiki/Public-domain-equivalent_license
Maybe CC0 or 0BSD are worth looking into.

Note the CC0 does explicitly not pass on patents.
>  No trademark or patent rights held by Affirmer are waived, abandoned, surrendered, licensed or otherwise affected by this document.

This might or might not be what you want.

0BSD does implicitly exhaust patent rights in my understanding.

maehne, 2019-08-29:
Are you aware of the CERN Open Hardware Licence? Their [wiki](https://www.ohwr.org/project/cernohl/wikis/home) contains also links to other licenses for open hardware projects as well as to guidelines and comparisons. There's also a [video](https://peertube.f-si.org/videos/watch/6bbd9f81-0ae0-4194-b10a-72f3ba6b4467) of Tristan Gingold's presentation of the draft for CERN Open Hardware Licence version 2 held at [FSiC 2019](https://wiki.f-si.org/index.php/FSiC2019).

1138-4EB, 2019-08-29:
> I think this is wrong since there are several licenses that work without attribution:
> https://en.wikipedia.org/wiki/Public-domain-equivalent_license
> Maybe CC0 or 0BSD are worth looking into.
>
> Note [...]

As said, all the legal infrastructure around licensing is based on attribution. This includes the very specific kind of licenses that you mention, which "try" to attribute the authority to the 'public domain'. 'Public domain' means 'anyone'; 'anyone' is not 'no one'. In other words, 'Public-domain-equivalent' licenses do not require to attribute the author of a work, but they do attribute the copyright holder very explicitly. If they didn't, there would be no reason for these licenses to exist.

There is a very simple explanation for this: in some countries it is not legally possible for authors to disclaim the copyright of the tools/product they produce.

> This might or might not be what you wat.
> 
> 0BSD does implicitly exhaust patent rights in my understanding.
> 
> Open Cores recommends LGPL https://cdn.opencores.org/downloads/opencores_coding_guidelines.pdf but i would hope for something more permissive.

If this is an issue to request that the project is donated to the public domain, I think you should do it explicitly, by arguing why you consider it less harmful or more desirable than other alternatives.

For my use case, there are lots of different licenses that might fit. Hence, I'm not going to push to either. I'm ok with any open source, GPL-alike, BSD-alike or public domain license.

@maehne, thanks for the references. I explicitly mentioned both the license and the presentation in my first comment above. However, I did not link the video. Precisely, I think that this project can wait an see how OHLv2 looks like before deciding to change the license, in case @antonblanchard is actually considering to do so.

freemin7, 2019-08-31:
You summarized the point of a 'Public-domain-equivalent' licenses well.
> If this is an issue to request that the project is donated to the public domain, I think you should do it explicitly, by arguing why you consider it less harmful or more desirable than other alternatives.

I do not want to convince anyone. I just wanted to make people aware that these licenses exist. I find it wrong to include things in a contract/license (like "attribution") when there is no will to enforce it. This is usually the case in Open Hardware.

aubindetrez, 2022-07-17:
https://github.com/antonblanchard/microwatt/blob/35e0dbed345011bafb651fa7d168de550e6fd6e7/LICENSE#L7-L8

That was in 2019, should the license be updated now?

Note: The End User license Agreement contains:
> 1.2 OPF grants to Recipient the right to license Recipient Power ISA Cores under the Creative Commons Attribution
> 4.0 license.
> 1.3 OPF grants to Recipient the right to sell or license Recipient Power ISA Cores under independent terms that are
> consistent with the rights and licenses granted under this Agreement. As a condition of the license grant under this
> section 1.3, the Recipient must either provide the Power ISA with this Agreement to the downstream recipient, or
> provide notification for the downstream recipient to obtain the Power ISA and this Agreement to have appropriate
> patent licenses to implement the Power ISA Core as a Power Compliant Chip. It is clarified that no rights are to be
> granted under this Section 1.3 beyond what is expressly permitted by this Agreement.

Also: Do you know if you are allowed to distribute a copy of the Open PowerISA on your repository? (I am asking because I am implementing a Power ISA core as a "toy project" and it would be nice to distribute the `.pdf` together with the implementationl)