antonblanchard/microwatt #15: GHDL build on native powerpc64le requires explicit LLVM backend selection

justinlynn, 2019-09-01:
Debian doesn't build ghdl for powerpc64/le targets yet as package build results in incorrect mcode selection on powerpc64/le architectures. Manual specification of `./configure --with-llvm=$(which llvm-config)` results in working native GHDL build which can be used to simulate microwatt according to README instructions natively on powerpc64/le. Suggest modifying README to include instructions and workaround lack of GHDL package availability in powerpc64/le debian and debian derived distros.

antonblanchard, 2019-09-03:
Thanks @justinlynn, yes we should add something in the README.

antonblanchard, 2019-09-04:
@justinlynn would you be able to summarze the steps needed on Debian? Can we use the distro llvm here?

justinlynn, 2019-09-04:
@antonblanchard More than happy to make a PR with the needed changes for Debian, etc. to save you the trouble of translating. As to system LLVM - Yes, I've had success in building GHDL master with Ubuntu Disco's LLVM development packages. If you'd like I can work on streamlining the build/simulation experience with some of the tooling (docker, etc) I've put together and submit as small separated PRs - thoughts?

justinlynn, 2019-09-04:
For anyone reading this in the future the steps required are:

1. Clone `https://github.com/ghdl/ghdl.git` into a local dir.
2. Use apt to install `llvm-8-dev`, assuming you already have a basic toolchain and such installed (`build-essentials`)
3. Move into the GHDL checkout.
4. Run `./configure --with-llvm-config=/usr/bin/llvm-config-8`, using `--prefix` to some location you want to install GHDL, if desired (for example `$HOME/.local`).
5. Fix any issues raised by configure, repeat from 4 until clear.
6. Run `make && make install`
7. Assuming where you installed GHDL is on your $PATH, microwatt's makefile should now work.

antonblanchard, 2019-09-05:
@justinlynn sounds great! We've had a couple of people trip over the issues with Debian and Ubuntu, so something in the README would be very useful. Also interested to see your tooling.

I think @sharkcz was looking at ppc64le distro issues too.

sharkcz, 2019-09-05:
Yes, things work out of the box on Fedora with packaged `ghdl`.

classilla, 2019-09-05:
Cool. I'll have to try that now.

sharkcz, 2019-09-05:
for the record there is a new ghdl in Rawhide, I don't plan to push this build into the stable releases (like F-30), but will make it available in https://copr.fedorainfracloud.org/coprs/sharkcz/danny/

mikey, 2020-06-11:
@justinlynn did you want to contribute those instructions to README.md and create a PR?

justinlynn, 2020-06-12:
Yes, I can do that shortly.

mikey, 2020-06-12:
No worries, I was just cleaning up the issues and wanted to close this out. Thanks @justinlynn

justinlynn, 2021-06-23:
I am so sorry I let this go for so long :/ At the very least for Ubuntu 20.10 or higher with more recent ghdl builds this now works out of the box. Closing this issue.

justinlynn, 2021-06-23:
Actually, no, that's not the case. My mistake. I'll take care of this ASAP.