antonblanchard/microwatt #111: just to play : using windows wsl and ubuntu , wuilding ghdl and running micropython

titof49, 2019-10-18:
its relatively slow but it work using Windows wsl , Ubuntu and building ghdl with llvm  ;-> using your tutorial to test.

[image]
sudo apt-get install --no-install-recommends git cmake ninja-build gperf   ccache dfu-util device-tree-compiler wget   python3-pip python3-setuptools python3-tk python3-wheel xz-utils file   make gcc gcc-multilib
   54  sudo apt-get install gcc-4.8-powerpc-linux-gnu g++-4.8-powerpc-linux-gnu binutils-4.8-powerpc-linux-gnu
   55  sudo apt-get install gcc-powerpc64le-linux-gnu
   56  sudo apt-get install g++-powerpc64le-linux-gnu
   57  sudo apt-get install binutils-powerpc64le-linux-gnu
   58  sudo apt-get install binutils-powerpc64le-linux-gnu-dbg
   61  sudo apt-get install binutils-powerpc64le-linux
   64  sudo apt-get install gcc-4.8-powerpc-linux-gnu g++-4.8-powerpc-linux-gnu binutils-4.8-powerpc-linux-gnu
   75  sudo apt-get install ghdl
   85  sudo apt-get install ghdl gtkwave
  118  sudo apt-get install LLVM
  119  sudo apt-get install llvm
  120  sudo apt-get install llvm-8

and for ghdl

https://gist.github.com/mathieucaroff/73ccbd30638d9b37b7129a7b7b8d7726

with these parameters :

 ./configure --prefix=/usr/local --with-llvm-config=/usr/bin/llvm-config-8

eine, 2020-01-17:
@titof49, which version of WSL are you using? Is it WSL2? Last time we tried, GHDL would not work on WSL1: ghdl/ghdl#551. Maybe it is because of the backend?

mikey, 2020-06-11:
I'm closing this as it's not an issue AFAICT.

closed 2020-06-11 by mikey