antonblanchard/microwatt #172: Building for ECP5 FPGA fails

carlosedp, 2020-05-11:
When building for ECP5 FPGA using Docker images, I get the following error:

```
❯ make -f Makefile.synth
docker run --rm -v /Users/cdepaula/repos/microwatt:/src:z -w /src ghdl/synth:beta yosys -m ghdl -p "ghdl --std=08 -gMEMORY_SIZE=8192 -gRAM_INIT_FILE=hello_world/hello_world.hex -gRESET_LOW=true -gCLK_INPUT=12000000 -gCLK_FREQUENCY=12000000 fpga/soc_reset.vhdl fpga/clk_gen_bypass.vhd decode_types.vhdl common.vhdl wishbone_types.vhdl wishbone_debug_master.vhdl wishbone_arbiter.vhdl cache_ram.vhdl utils.vhdl plru.vhdl helpers.vhdl dcache.vhdl core_debug.vhdl fetch1.vhdl fetch2.vhdl register_file.vhdl insn_helpers.vhdl multiply.vhdl divider.vhdl logical.vhdl crhelpers.vhdl countzero.vhdl rotator.vhdl ppc_fx_insns.vhdl execute1.vhdl decode1.vhdl cr_file.vhdl writeback.vhdl loadstore1.vhdl icache.vhdl cr_hazard.vhdl gpr_hazard.vhdl control.vhdl decode2.vhdl core.vhdl fpga/pp_fifo.vhd fpga/pp_soc_uart.vhd dmi_dtm_dummy.vhdl fpga/main_bram.vhdl wishbone_bram_wrapper.vhdl soc.vhdl fpga/toplevel.vhdl -e toplevel; synth_ecp5 -json microwatt.json"

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Wolf <claire@symbioticeda.com>          |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.9+2406 (git sha1 UNKNOWN, clang 7.0.1-8 -fPIC -Os)

-- Running command `ghdl --std=08 -gMEMORY_SIZE=8192 -gRAM_INIT_FILE=hello_world/hello_world.hex -gRESET_LOW=true -gCLK_INPUT=12000000 -gCLK_FREQUENCY=12000000 fpga/soc_reset.vhdl fpga/clk_gen_bypass.vhd decode_types.vhdl common.vhdl wishbone_types.vhdl wishbone_debug_master.vhdl wishbone_arbiter.vhdl cache_ram.vhdl utils.vhdl plru.vhdl helpers.vhdl dcache.vhdl core_debug.vhdl fetch1.vhdl fetch2.vhdl register_file.vhdl insn_helpers.vhdl multiply.vhdl divider.vhdl logical.vhdl crhelpers.vhdl countzero.vhdl rotator.vhdl ppc_fx_insns.vhdl execute1.vhdl decode1.vhdl cr_file.vhdl writeback.vhdl loadstore1.vhdl icache.vhdl cr_hazard.vhdl gpr_hazard.vhdl control.vhdl decode2.vhdl core.vhdl fpga/pp_fifo.vhd fpga/pp_soc_uart.vhd dmi_dtm_dummy.vhdl fpga/main_bram.vhdl wishbone_bram_wrapper.vhdl soc.vhdl fpga/toplevel.vhdl -e toplevel; synth_ecp5 -json microwatt.json' --

1. Executing GHDL.
soc.vhdl:207:24: unit "xics" not found in library "work"
ERROR: vhdl import failed.
gmake: *** [Makefile.synth:69: microwatt.json] Error 1
```

carlosedp, 2020-05-13:
Adding `xics.vhdl` to `VHDL_FILES`, now gives me:

```
...
Removed 0 unused cells and 58406 unused wires.

2.49. Executing AUTONAME pass.
gmake: *** [Makefile.synth:61: microwatt.json] Error 137
```

carlosedp, 2020-05-13:
One error at a time. After updating yosys and nextpnr containers, I'm getting this at nextpnr-ecp5:

```
docker run --rm -v /Users/cdepaula/repos/microwatt:/src:z -w /src ghdl/synth:nextpnr-ecp5 nextpnr-ecp5 --json microwatt.json --lpf constraints/ecp5-evn.lpf --textcfg microwatt_out.config --um5g-85k --freq 12 --package CABGA381

Info: Logic utilisation before packing:
Info:     Total LUT4s:     43536/83640    52%
Info:         logic LUTs:  40316/83640    48%
Info:         carry LUTs:   2212/83640     2%
Info:           RAM LUTs:    672/41820     1%
Info:          RAMW LUTs:    336/20910     1%

Info:      Total DFFs:     24566/83640    29%

Info: Packing IOs..
Info: pin 'ext_clk$tr_io' constrained to Bel 'X63/Y0/PIOA'.
Info: pin 'ext_rst$tr_io' constrained to Bel 'X0/Y92/PIOD'.
Info: pin 'uart0_rxd$tr_io' constrained to Bel 'X0/Y92/PIOB'.
Info: pin 'uart0_txd$tr_io' constrained to Bel 'X0/Y92/PIOC'.
Info: Packing constants..
Info: Packing carries...
Info: Finding LUTFF pairs...
Info: Packing LUT5-7s...
Info: Finding LUT-LUT pairs...
Info: Packing paired LUTs into a SLICE...
Info: Packing unpaired LUTs into a SLICE...
Info: Packing unpaired FFs into a SLICE...
Info: Generating derived timing constraints...
Info: Promoting globals...
Info:     promoting clock net ext_clk$TRELLIS_IO_IN to global network
Info: Checksum: 0x57f781d5

Info: Annotating ports with timing budgets for target frequency 12.00 MHz
ERROR: cell type '$assert' is unsupported (instantiated as 'soc0.processor.register_file_0.4565')
0 warnings, 1 error
gmake: *** [Makefile.synth:64: microwatt_out.config] Error 255
```

mikey, 2020-05-13:
@carlosedp Thanks for reporting. Sorry I've not had time to look at this but I'll get to it eventually.

antonblanchard, 2020-05-19:
@carlosdep Thanks. This is being worked in https://github.com/YosysHQ/yosys/issues/2068. We also hit https://github.com/YosysHQ/yosys/issues/2065, and the fix just went in. We promise to add some Travis CI around this soon!

eine, 2020-06-11:
@carlosedp, the docker images used by the Makefile in this repo are maintained in [ghdl/docker](https://github.com/ghdl/docker). I think it is important for you to know that these images are slightly lagging behind `master` branches of the different projects/tools. As you can see in the README, the synchronization between repos in org [ghdl](https://github.com/ghdl) was recently implemented. As a result, all the images should be updated within a few hours after changes are pushed to ghdl/ghdl. However, yosys and nextpnr are updated weekly.

In the future, I'd like to split non-ghdl tools to SymbiFlow (see [eine/symbiflow-containers](https://github.com/eine/symbiflow-containers)) and set up cross-triggers between Yosys/nextpnr and SymbiFlow. However, the proposal is not ready yet.

As commented by @antonblanchard, CI was added (and it's being improved, #201). Hence, CI should let you/us know when the merged PRs in yosys do arrive to the released images.

mikey, 2020-06-25:
@carlosedp can you try #209 ?

mikey, 2020-07-07:
@carlosedp upstream should build now. The image doesn't work yet but I should have that soon.

mikey, 2020-07-07:
Fixed in #209

closed 2020-07-07 by mikey