antonblanchard/microwatt #215: Reads from the main memory can lock up the core

EMue, 2020-06-27:
I've been experimenting with Microwatt, attempting to get it to run on a Xilinx Zynq 7000 embedded SoC. I'm currently in the process of hooking up the DDR3 SDRAM.

While debugging, I've noticed that certain reads from main memory (0x0...), aliased to the memory the firmware is loaded from, lead to unexpected behavior. This can also be reproduced in simulation.

I've created a [sample program](https://github.com/EMue/microwatt/blob/experiments/read_test/read_test.c) to illustrate the issue. When the program is linked at 0x0, and is loaded from the block RAM, the core appears to hang indefinitely when reading a quad word from address 0x00001400 (text segment). This address probably depends on the exact compiler output.

ozbenh, 2020-06-29:
This is a software bug  :-)

First I must say, we should really put some basic exception vectors in our test programs, link them with the small libc we have,
and clear bss & setup a sane stack :)

So what's happening is that your program is taking a page fault, which hangs because we don't have proper exception handling.

The reason it is taking a fault is because, I think (from the logs), we have a collision between a cache-inhibited load and the L1D cache, which is detected and reported as an access error.

You are using readq() which uses cache-inhibited real mode instructions, but you are doing so on a range of memory that is accessed via the caches, I think in that case the collision is on the stack.

EMue, 2020-06-30:
Thank you for the clarification.

Seems that I'll have to read the specification and learn what these instructions actually do, and how exceptions are to be handled. :)
In the meantime, am I correct that the I/O functions are supposed to be used mainly for peripheral access? Or is it sufficient to access the specific BRAM/DRAM segments cache inhibited, while limiting "normal" reads and writes to the aliased main RAM (this appears to work OK on first glance).

Some basic examples for bare metal programming would indeed be helpful. Other than that, feel free to close this issue if you like.

ozbenh, 2020-06-30:
> Thank you for the clarification.
> Seems that I'll have to read the specification and learn what these
> instructions actually do, and how exceptions are to be handled. :)
> In the meantime, am I correct that the I/O functions are supposed to
> be used mainly for peripheral access? 

Yes. When the core is operating with the MMU on, you can (and should)
use normal load/store instructions, and the MMU attributes will define
whether a given page is cachable or not.

But when the MMU is off, which is the case with your test program, the
way to do IO accesses without caches is to use those special
instructions.

Architecturally they are only supposed to be available in "hypervisor
real mode", that said Microwatt is a bit more flexible.

> Or is it sufficient to access the specific BRAM/DRAM segments cache
> inhibited, while limiting "normal" reads and writes to the aliased
> main RAM (this appears to work OK on first glance).

You can do what you want as long as you don't have a collision between
a cache-inhibited access and something in the cache. Microwatt will
barf if it detects such a thing.

That said, on POWER, it's generally recommended to avoid accessing
normal memory cache inhibited.

> Some basic examples for bare metal programming would indeed be
> helpful. Other than that, feel free to close this issue if you like.

Cheers,
Ben.

mikey, 2020-07-07:
@EMue @ozbenh can I close this as not a bug?

closed 2020-07-08 by mikey