antonblanchard/microwatt #217: Vivado Implementation has negative slack

ChanonTonmai, 2020-07-01:
I have just run the microwatt with fusesoc on arty_a7-35. Then, I open the microwatt_0.xpr in vivado gui mode. I found that there are critical warning about the timing. It said that the design failed to meed the timing requirement.
[image]
The ram initial is the hello_world.hex which is in this git. 

Moreover, I saw the utilization is quite large for arty7.
[image]
Is there any suggestion for solve the slack and lean this core?

EMue, 2020-07-01:
It appears that with these smaller FPGAs, especially at utilization this high, something needs to give.

The following might help:
- Reduce the core clock frequency. 50 Mhz is much easier to achieve than 100.
- Try different implementation strategies in Vivado. "Performance-ExtraTimingOpt" might help. Implementation will take longer though.
- Experiment with reducing the block RAM size.

Also make sure you're operating on the current revision of Microwatt. There have recently been some timing improvements.

Without the latter two optimizations, I've never gotten Vivado to achieve 100 Mhz on a Zynq 7010. With both enabled, it just about seems to work. 50 Mhz, however, has never given me any trouble.

ChanonTonmai, 2020-07-01:
As an information, firstly, I change the strategies in Vivado to "Performance-ExtraTimingOpt". The total negative slack is decrease. 
[image]
Then, when I change the sys_clk_pin to 50MHz . The negative slack is disappear but it has total pulse width slack. I don't know what the tpws mean but I think it is not the problem. 
[image]

I going to close this issue. Moreover, I saw that you have been implement this core on Zynq. Is the xdc file is the one that need to change? I have Zybo (7010) and Pynq (7020). If you use the same board, could you share the xdc file. 

Thank you

closed 2020-07-01 by ChanonTonmai

PrithvirajChauhan1, 2021-05-28:
Hey can you help me out in implementing this on vivado i am unable to do that...