antonblanchard/microwatt #221: Add Processor Version Register (PVR) of 0063

mikey, 2020-07-06:
microwatt has been allocated by the PVR gods a PVR version of 0063. Implement this.

classilla, 2020-07-07:
There are PVR gods? I wonder what their shrine would look like.

mikey, 2020-07-07:
I asked inside IBM and someone with an authoritative voice said we could have this one. Sounded god like to me ;-)

FWIW 0063 is in a non-IBM range of versions.
0060 and 0061 were this processor https://en.wikipedia.org/wiki/Exponential_Technology
0062 was allocated to a Hitachi powerpc processor.

classilla, 2020-07-07:
Exponential Technology! I remember the X704 was supposed to be the next great thing in Power Macs, and it was a hot mess.

iamjpn, 2020-07-07:
I had a go at this:
https://github.com/antonblanchard/microwatt/pull/222

mikey, 2020-07-07:
Looks good @iamjpn . 

Did you want to write a console test for it? Maybe extend test/misc? 

Writing a test forth will be rejected! :-P

iamjpn, 2020-07-07:
Yeah I'll add a test for it.
Also I guess I should check for Problem space and raise a Privileged Instruction type Program Exception?

mikey, 2020-07-07:
Yes please. You can add that part of the test to tests/privileged.

closed 2020-07-09 by mikey