antonblanchard/microwatt #224: Add instruction breakpoint

mikey, 2020-07-08:
Add a instruction breakpoint to microwatt. When the PC hits this value, take a trace exception at 0xD00.

The ISA defines the CIABR which is the "completed" instruction. This means take the exception **after** the instruction is run which is not ideal from a software perspective. An implementation that takes the breakpoint **before** the instruction is executed would be more useful but we'd have to call it an IABR not CIABR.

mikey, 2020-07-08:
Extra credit for adding single stepping and/or branch tracing via MSR[TE].

iamjpn, 2020-07-09:
I've got a bit of a WIP here: https://github.com/iamjpn/microwatt/tree/ciabr

mikey, 2020-07-13:
> Doesn't actually seem to complete first the instruction though.

What do you mean by that? @iamjpn

iamjpn, 2020-07-13:
> > Doesn't actually seem to complete first the instruction though.
> 
> What do you mean by that? @iamjpn

What I mean is that currently the exception is taken before the instruction is completed.
```
breakpoint:
	addi	%r4, %r4, 1
	b	1f // should be skipped
	//cmpwi	%r4, 1
	//bne	1f
```
That's why I commented out that cmpwi in the test.

mikey, 2020-07-14:
As mentioned in the first comment, it would be useful to have it that way but it's not arch compliant.

iagocaran, 2023-01-10:
I reimplemented @iamjpn work for a newer version of the code and wanted to make it arch compliant, but I'm having trouble getting the completed instruction address instead of the next instruction address. 

https://github.com/iagocaran/microwatt/commit/6592df0e4e6f6af3c180f2a9c7e3d10e81c7ecc4