antonblanchard/microwatt #231: firmware sources

sharkcz, 2020-07-10:
Looks to me that `litedram/gen-src/sdram_init/` is missing some source files, it can't be rebuilt.
````
[dan@talos sdram_init]$ make
Makefile:3: variables.mak: No such file or directory
make: *** No rule to make target 'variables.mak'.  Stop.
````

The other question is whether shouldn't be a separate project for Microwatt's firmware :-)

mikey, 2020-07-13:
@ozbenh ping

ozbenh, 2020-07-13:
This isn't how you build it :-)

It's built by litedram/gen-src/generate.py

The build process uses a combination of sources in LiteX itself and the stuff in litedram/gen-src/sdram_init. Since LiteX tends to change, it's a fairly fragile process which is why I ship pre-built binaries.

(For examples it won't build with current LiteX unless you use my litedram branch)

The main reason for that is that the required inits for LiteDRAM rely on data generated along with the .v core which might change as LiteX evolves.

Now to clean things up a bit and make it easier to have our own firmware independant of LiteDRAM, I want to eventually change this to generate a litedram.a as a library which only contains the DRAM inits (and is pre-generated along with the .v cores), and have elsewhere a more common firmware that can optionally link against the above. But I haven't done that yet.

ozbenh, 2020-07-13:
(Mikey, you can probably close this)

mikey, 2020-07-13:
@ozbenh should we just remove that whole sdram_init dir then? Or at least put a README/Build instructions in there describing this so people don't trip up on this again?

ozbenh, 2020-07-13:
Why remove it ? it would make things more messy ... it contains the bits that generate.py uses (along with LiteX) to generate the inits ...

But yes, adding a README of some sort makes sense

ozbenh, 2020-07-13:
we should also remove the copy of bin2hex in there :)

ozbenh, 2020-07-13:
I was thinking for the Makefile, we could actually have generate.py invoke a specific non-default target and have the default one just print a message.. I can try to look into these later this week but feel free to beat me to it