antonblanchard/microwatt #234: Provide git ID of microwatt source in a syscon register

paulusmack, 2020-07-16:
It would be cute to be able to identify the source code from which a particular microwatt cpu implementation was built. If the build process could look for the git SHA1 ID of the current head and include that as a constant value in the build, hooked up to the syscon framework so it could be read by MMIO, that could be useful. We would probably want (say) 32 bits of SHA1 plus a dirty flag.

Then the boot code could be enhanced to print that at startup along with the Soc signature, features, etc.

mikey, 2020-07-22:
Getting hello world to print it would be nice too.

sharkcz, 2020-09-02:
Implemented in PR #243

mikey, 2022-08-30:
Merged in PR #400 ! Woot. Thanks @sharkcz

closed 2022-08-30 by mikey