antonblanchard/microwatt #246: Compatibility with compiled Micropython

shivampotdar, 2020-09-24:
I see that the prebuilt image in this repository is based on MicroPython v1.12-571 (some intermediate version I guess?)

I tried compiling micropython using GCC cross-compiler on Ubuntu 20.04 (with both buildroot download and gcc9.3 based from debian repositories)
I attempted using the current master branch, v1.13 and v1.12 from the micropython repo.
In all the cases, although I can compile without errors, when I run `core_tb`, I am not able to issue any inputs in the terminal from the keyboard. This does not happen with the prebuilt image.

I am not sure how exactly shall I debug this. But I would be happy to help and provide more details if neeeded.

Thank you.

paulusmack, 2020-11-03:
This sounds like your image is trying to access the potato UART but core_tb has the 16550-style UART.

titof49, 2021-09-05:
Hi @paulusmack Paul @antonblanchard , i had the same issue.

I saw that you can set the 16550-style UART.
 but from my previous tests it wasn't functionnal.
 i'll check it with the latest version of mycropython for powerpc.

titof49, 2021-09-05:
[Solved]

@paulusmack @antonblanchard @shenki *

to correct the issue, you can change in the makefile by changing :

# potato or lpc_serial
UART ?= potato

by 

UART ?= lpc_serial

after you can rebuild and recreate the link 

enjoy again ;->>

below the full modifications:

titof49@pcdev:~/uwatt/micropython/ports/powerpc$ diff Makefile Makefile.old
10c10
< UART ?= lpc_serial
---
> UART ?= potato

antonblanchard, 2021-09-25:
Thanks @titof49! We should update micropython to use the 16550 all the time, and eventually get rid of the potato UART in microwatt.

iagocaran, 2022-05-24:
Instead of changing the makefile, you can compile like this:

`make -j$(nproc) UART=lpc_serial`

167rgc911, 2023-01-10:
In the latest code this seems to be broken again:

(1) potato (default build of micropython):

==========
rgc:~/i/microwatt$ ./core_tb > debug_log
MicroPython v1.19.1-782-g699477d12 on 2023-01-10; bare-metal with POWERPC
Type "help()" for more information.
>>>
==========

- micropython seems to output the debug string properly
- but user is unable to perform any input

(2) UART (lpc_serial in micropython):

==========
rgc:~/i/microwatt$ ./core_tb > debug_log

==========

- micropython can't even output any debug string

AFAIK lpc_uart_init() is 
https://github.com/micropython/micropython/blob/1f371947309c5ea6023b6d9065415697cbc75578/ports/powerpc/uart_lpc_serial.c#L100

using a different base address now (LPC_UART_BASE).
https://github.com/micropython/micropython/blob/1f371947309c5ea6023b6d9065415697cbc75578/ports/powerpc/uart_lpc_serial.c#L39

titof49, 2023-01-10:
You must change uart mode see #246 issue on git mocrowatt

https://github.com/micropython/micropython/blob/1f371947309c5ea6023b6d9065415697cbc75578/ports/powerpc/uart_lpc_serial.c#L100
https://github.com/micropython/micropython/blob/1f371947309c5ea6023b6d9065415697cbc75578/ports/powerpc/uart_lpc_serial.c#L39