antonblanchard/microwatt #248: ghdl:error: unknown warning identifier

brunomaximom, 2020-10-06:
Trying to build microwatt on Debian with make and I get
ghdl:error: unknown warning identifier

Using ghdl-nightly from https://github.com/ghdl/ghdl/releases built from source code, but I get the same error with 0.37.

cr1901, 2021-02-21:
For anyone else who runs into this error, this fixed the issue for me:

You need to recompile `ghdl` with the `--with-llvm-config` option. Example:

```
../ghdl/configure --prefix=$PREFIX --enable-libghdl --with-llvm-config
```

In addition, this _must_ be a clean build. I.e. run `make distclean` first in your build directory. If you don't, you may get loads of `undefined reference to `__gnat_last_chance_handler'` at link time.

umarcor, 2021-07-21:
Currently, microwatt can only be used with GHDL's LLVM or GCC backends, not mcode. That is because how the Makefile was written for integrating the simulation RAM model written in C, as well as the console and the JTAG socket. Precisely, mcode is a built-in in-memory backend which does not support compiling additional sources or linking additional objects. However, the Makefile relies on `soc_sim_obj_files` being compiled to `*.o` and then passed as a `soc_sim_link` to the GHDL elaboration.

The solution for supporting mcode would be to compile `soc_sim_obj_files` into a shared library (`*.so` or `*.dll`) and let VHPIDIRECT get the foreign functions from there. In fact, that same procedure should work for LLVM or GCC too.

References:

- https://ghdl.github.io/ghdl-cosim/vhpidirect/linking.html
- https://ghdl.github.io/ghdl-cosim/vhpidirect/dynamic.html#loading-foreign-objects-from-within-a-simulation
- https://ghdl.github.io/ghdl-cosim/vhpidirect/examples/quickstart.html#sin-from-libmath
- https://github.com/ghdl/ghdl/issues/1777

Note that it is already told in the README that LLVM or GCC need to be used:

> https://github.com/antonblanchard/microwatt#simulation-using-ghdl
>
> Microwatt requires ghdl to be built with the LLVM or gcc backend, which not all distros do (Fedora does, Debian/Ubuntu appears not to). ghdl with the LLVM backend is likely easier to build.

robyinno, 2022-12-01:
with Linux Mint 21 I have installed all the ghdl packages except ghdl-mcode and error was solved 
```
apt search ghdl
i   ghdl                            - compilatore e simulatore di VHDL          
i A ghdl-common                     - VHDL compiler/simulator (common files)    
i   ghdl-gcc                        - VHDL compiler/simulator (GCC backend)     
i   ghdl-llvm                       - VHDL compiler/simulator (LLVM backend)    
p   ghdl-mcode                      - VHDL compiler/simulator (mcode backend) 
```