antonblanchard/microwatt #270: Can Microwatt run on a Raspberry Pi Pico?

MailYouLater, 2021-01-22:
I'm not very familiar with PowerPC/POWER/OpenPower, or with microcontrollers, FPGAs or hardware design in general, so Microwatt is new to me in basically every way and I don't know if this even makes any sense, but... I'm wondering, if I wanted to experiment with the Microwatt project, would I be able to use a Raspberry Pi Pico to simulate an OpenPower ISA processor on real hardware, similar to how you could use Microwatt on an FPGA? (or am I just making connections where there aren't any?)

As I said, I don't really know enough about any of the technologies used in this project to even know if this makes any sense, but it seems micropython is used by the Microwatt project, and when I saw on the Raspberry Pi website that the Pico is "Programmable in C and MicroPython", I thought of Microwatt and wondered if it was possible to have it operate as a small OpenPower CPU on via the Microwatt project.

Please let me know if this is possible, and if not, what the technical reasons are.  Thank you!

mikey, 2021-01-28:
Thanks for your inquiry @MailYouLater 

It's not really possible to run microwatt on the Raspberry Pi Pico. That board is just a tiny ARM CPU with very little memory. You can run small bare metal ARM applications (like micropython, or C) but nothing much else. 

You'd be better off using your laptop/desktop to simulate microwatt using GHDL.

closed 2021-01-28 by mikey

MailYouLater, 2021-02-03:
@mikey Hi.  The way you worded your response, I'm not sure if it's actually not possible, or if it's just extremely impractical.  Can you possibly tell me which it really is?  Is the Raspberry Pi Pico actually _incapable_ of simulating the Microwatt CPU (and if so what makes it impossible?) or is it technically possible, but not worth bothering with? (in which case, what about the Pico makes it so infeasible?)

mikey, 2021-02-03:
With only 256KB of memory in the Raspberry Pi Pico, I would say it's impossible. @MailYouLater

MailYouLater, 2021-02-03:
Alright.  Thanks for specifying.