antonblanchard/microwatt #300: Verilator and Yosys support broken

zyedidia, 2021-07-14:
I've been trying to use the `microwatt.v` and `microwatt-verilator` targets in the makefile, but it seems like both are broken. The `microwatt.v` file produced by Yosys generates invalid Verilog, which Yosys itself cannot read back. The issue seems to be caused by Yosys generating a few statements of the form `assign "" = ""`. Should I report this directly to Yosys?

After removing those lines, Yosys can read the file in but Verilator still gives the error:

```
%Error: microwatt.v:203:3: Cannot find file containing module: 'EHXPLLL'
```

along with some warnings (`LITENDIAN`, and `PINMISSING`) though perhaps these can be ignored.

Are these targets supposed to be working, or are they still in-progress?

I'm using versions built fairly recently from source: Yosys 0.9+3672, Verilator 4.203 devel rev v4.202-56-g1f331bd9, and GHDL 2.0.0-dev (1.0.0.r333.g17ddaa0d).

umarcor, 2021-07-21:
@zyedidia it seems that the Yosys -> Verilog -> Verilator -> Simulation workflow is not used/test in CI. The generation of the verilog source is tested only: https://github.com/antonblanchard/microwatt/blob/master/.github/workflows/test.yml#L88.

However, using git blame as a reference (https://github.com/antonblanchard/microwatt/blame/f9654428ff28744dcf1ad4e7fe43604415689f36/Makefile#L196), it seems that @antonblanchard is verilator, because it was modified some months ago: https://github.com/antonblanchard/microwatt/pull/286/files.

According to https://github.com/antonblanchard/microwatt/commit/df8e1ca2a7a4c59ebaf4fb09a485a641c99c7c80, you need to set `FPGA_TARGET=verilator`. That would explain why you are having problems with EHXPLL (the default target is the OrangeCrab/ECP5: https://github.com/antonblanchard/microwatt/blob/master/Makefile#L144).

antonblanchard, 2021-09-24:
Thanks for investigating @umarcor! You are correct in that `FPGA_TARGET=verilator` needs to be set. This is confusing, and the Makefile is a bit of a mess, but I had hoped we'd be able to drive `ghdl` and `yosys` with `fusesoc`. Hopefully one day.

We now have a CI test that builds a verilator binary and runs micropython.

umarcor, 2021-09-24:
@antonblanchard, precisely microwatt is one of the projects I'm using as a reference for [OSVB: Core](https://umarcor.github.io/osvb/apis/core.html) and when working on the ProjectModel in EDA² ([edaa-org.github.io](https://edaa-org.github.io)). The Makefile and `.core` files in this repository are rather challenging, so they are a pretty strong test for any project trying to deal with "HDL and EDA tooling".

That is also related to VUnit, which was contributed here by @LarsAsplund a few months ago. We are looking into some nicer integration of VUnit with FuseSoc/Edalize. That's actually what the pyCAPI proof-of-concept in OSVB is about.