antonblanchard/microwatt #306: Adding FPGA board support

nyaasen, 2021-08-05:
I am interested in adding support for the Basys3 FPGA that uses the same exact fpga part as the Cmod A7-35 fpga.

I have added a file in fpga/basys3.xdc that has the fixed pin numbers. I have also added the configuration in microwatt.core as basically a copy of cmod where it appears both times, with the exception of the first file pointing to fpga/basys3.xdc.

What other changes would I have to make? I believe I am missing something, but fusesoc gives no errors and the board does seem to receive a bitstream, but I have not been able to obtain serial out.

Thanks.

I can attach code if that would help

umarcor, 2021-08-07:
@nyaasen you might get feedback from fusesoc users/developers in [gitter.im/librecores/fusesoc](https://gitter.im/librecores/fusesoc). Even though fusesoc gives no errors, checking the logs might help confirm that the bitstream generation is correct.

/cc @olofk

olofk, 2021-08-10:
Thanks for the notice @umarcor. Checking the logs would be my first thing to do as well. If there is nothing obvious wrong, then you could try adding a blinking LED or something to check that the clock and reset behaves as expected. And if you have your modifications published somewhere it will be easier to see if there are anything wrong with the constraints or core description file

nyaasen, 2021-08-15:
Hi umarcor and olofk, 

I have uploaded my changes here: https://github.com/nyaasen/microwatt
I went ahead and asked on the gitter page for fusesoc about the log files and will post back here when I have more information. As for adding the blinking LEDs for the clock and reset, that's something I will have to figure out how to do.

Thanks you two

nyaasen, 2022-08-11:
I'm going to go ahead and close this issue as I was able to build it through LiteX for the Basys3.

Thanks

closed 2022-08-11 by nyaasen