antonblanchard/microwatt #355: Add support for NiteFury/LiteFury

antonblanchard, 2022-03-08:
There's interest in getting Microwatt going on the NiteFury/LiteFury boards. Other than the usual constraint and top level files, Luke points out the lack of a UART adds another challenge.

mithro, 2022-03-09:
According to the [`sqrl_acron.py` file in LiteX](https://github.com/litex-hub/litex-boards/blob/master/litex_boards/platforms/sqrl_acorn.py);
> ```python
> # The Acorn (CLE-101, CLE-215(+)) are cryptocurrency mining accelerator cards from SQRL that can be
> # repurposed as generic FPGA PCIe development boards:
> # - http://www.squirrelsresearch.com/acorn-cle-101
> # - http://www.squirrelsresearch.com/acorn-cle-215-plus
> # The 101 variant is eguivalent to the LiteFury and 215 variant equivalent to the NiteFury from
> # RHSResearchLLC that are documented at: https://github.com/RHSResearchLLC/NiteFury-and-LiteFury.
> ```

LiteX also has a way to provide a UART over the PCIe bus (and also updating the bitstream).

Some random links;
 * https://twitter.com/SeanMBlakley/status/1459971651504852998?s=20
 * https://twitter.com/enjoy_digital/status/1462853379076628482?s=20
 * https://twitter.com/gbuzogany/status/1470127685103927303?s=20
 * https://twitter.com/enjoy_digital/status/1462876566531231751?s=20
 * https://twitter.com/hansfbaier/status/1482977512879980545?s=20

[image]

mithro, 2022-03-09:
* https://twitter.com/enjoy_digital/status/1480427484671795201?s=20 -- `--uart-name=crossover`

[image]
[image]

mithro, 2022-03-09:
There seems to be a full set of instructions at https://teaandtechtime.com/litex-on-the-sqrl-acorn-fpga/

mithro, 2022-03-09:
Even more;
 * https://github.com/enjoy-digital/litex-acorn-baseboard
 * https://github.com/Victoromoniyi/gr-litex
 * https://github.com/SMB784/SQRL_quickstart

paulusmack, 2026-02-11:
This is addressed by commit dbb137437c90 ("acorn: Add support for the Acorn CLE 215+", 2020-07-08).

closed 2026-02-11 by paulusmack