antonblanchard/microwatt #363: Litedram no longer works on Genesys2 (and possibly NexysVideo and Nexys4DDR)

shingarov, 2022-06-05:
Commits 07f2edc / 780d6c7 port microwatt across an incompatible API change in litedram.  Unfortunately, newer litedram no longer works on Genesys2 (and possibly on NexysVideo and Nexys4DDR, as @fontamsoc reports in https://github.com/enjoy-digital/litedram/issues/297#issuecomment-1069258577 ).

As a temporary workaround, branch
https://github.com/shingarov/microwatt/tree/fix-dram
reverts to using the old API, so that one can use all the progress that happened in microwatt since August 2020 (when I last regenerated litedram for the newly-introduced Genesys2 port of microwatt), and still have memtest pass.

I am not sure what we should do about this, and the present Issue is more of a FYI about writing down what I observed and did to work around it for the time being, than a call to any particular action.  I'll also note that I have built today's tip of litex's master with the default vex cpu and it passes memtest.  So perhaps the problem is that there is more to migrating to the new API than changing a function name to sdram_init().  More investigation is required.

antonblanchard, 2022-08-08:
Thanks for the heads up @shingarov. We recently updated litedram in Microwatt, and Nexys Video was working before and after. Perhaps this is a Genesys2 only issue.

shingarov, 2022-09-18:
I don't have a Nexys Video to investigate what's going on there, but https://github.com/enjoy-digital/litedram/issues/281 still sits as "Open" and "Regression".  Maybe it's been fixed and it's just the status of the issue that is not updated; I'll try asking them.