antonblanchard/microwatt #386: System call exceptions aren't always executed in single step mode

antonblanchard, 2022-08-07:
It looks like we are not always executing system call exceptions when single stepping. We should see 10 system call exceptions and 23 single step exceptions. We only see 1 system call exception (a result of the second of the two back to back sc instructions).

```
/* Load an immediate 64-bit value into a register */
#define LOAD_IMM64(r, e)			\
	lis	r,(e)@highest;			\
	ori	r,r,(e)@higher;			\
	rldicr	r,r, 32, 31;			\
	oris	r,r, (e)@h;			\
	ori	r,r, (e)@l;

#define MSR_SE	(1UL << 10)

// r30 number of trace exceptions
// r31 number of system call exceptions

. = 0x0
	li	30,0
	li	31,0

	li	0,0x100
	mtsrr0	0

	mfmsr	0
	LOAD_IMM64(2, MSR_SE)
	or	0,0,2
	mtsrr1	0
	rfid

. = 0x100
	nop
	sc
	nop
	sc
	nop
	nop
	sc
	nop
	nop
	nop
	sc
	nop
	nop
	nop
	nop
	sc
	nop
	nop
	nop
	nop
	nop
	sc
	nop
	nop
	nop
	nop
	nop
	sc
	divd 0,0,0
	sc
	nop
	sc
	sc
	trap

. = 0x700
	attn
	
. = 0xc00
	addi	31,31,1
	rfid

. = 0xd00
	addi	30,30,1
	rfid
```

```
powerpc64le-linux-gnu-gcc -O2 -c testcase.S -o testcase.o
powerpc64le-linux-gnu-objcopy -O binary testcase.o testcase.bin
cp  testcase.bin main_ram.bin
./core_tb > out
```

...

```
register_file.vhdl:201:21:@4625ns:(report note): GPR30 0000000000000017
register_file.vhdl:201:21:@4625ns:(report note): GPR31 0000000000000001
```

antonblanchard, 2022-08-07:
Bisects to 813e2317bf1f1c10d988f660c0a4282da316a3b9 ("execute1: Restructure to separate out execution of side effects")

paulusmack, 2022-08-09:
It seems that for the first trace interrupt, SRR0 is set to 108, not 104, hence we go back to the second nop not the first sc.

closed 2022-08-09 by antonblanchard