antonblanchard/microwatt #394: Graph FPGA resource usage with over time

mikey, 2022-08-11:
I'd like to graph the usage of FPGA resource usage over time so we can easily see when things bloat.

When building microwatt.json, you get a report from yosys like this:
```
      CCU2C                        1881
      DP16KD                         54
      EHXPLLL                         1
      LUT4                        27427
      MULT18X18D                     32
      TRELLIS_DPR16X4               512
      TRELLIS_FF                   9646
```
I'd like to graph these parameters on each build (or daily) to see how they are changing.

I assume someone has done something similar for code bloat but I'm not sure what the best way/state of the art to do this is. 

https://github.com/jdjohn215/milwaukee-weather is a neat way to grab data and graph it each day, so something like that could work. We could do it as a separate git repo that runs regularly but open to better ideas.

mikey, 2022-08-11:
Referencing https://github.com/antonblanchard/microwatt/issues/302 that might help us.

mikey, 2022-08-25:
Graphing some benchmarks over time would also be useful

mithro, 2022-08-25:
FYI - https://github.com/mattvenn/logLUTs might be useful.

mithro, 2022-08-25:
https://github.com/leaflabs/xmsgs-tools might also be helpful too.

mikey, 2022-08-30:
@mithro I kinda got lutLUTS to work. Now I need to automate it
[image]