antonblanchard/microwatt #398: apparent bug in plru

programmerjake, 2022-08-19:
I [re-translated plru.vhdl to nmigen as part of writing a formal correctness proof](https://bugs.libre-soc.org/show_bug.cgi?id=909#c1), and discovered what looks like a bug to me (assuming I properly understand vhdl's semantics):

https://github.com/antonblanchard/microwatt/blob/f67b1431655c291fc1c99857a5c1ef624d5b264c/plru.vhdl#L36-L47

On line 41, `tree(node)` is used after multiplying `node` by `2`, which looks incorrect, it should use the value of `tree(node)` before `node` is modified in the current loop iteration, since that's what's assigned to `lru` and printed to the output when the `report` is un-commented.

In order for my translation to match the independent implementation I tested against, using the `node` before multiplying by `2` is necessary.

programmerjake, 2022-08-19:
as a side-note, `tree`'s last entry isn't used, it's 1 too large. This shouldn't cause issues.

mikey, 2022-08-24:
@ozbenh I think you wrote this code. Care to comment?

ozbenh, 2022-08-25:
I think @programmerjake is correct. I haven't touched any of this in a while and have been having difficulties with ghdl :-) So this is lightly tested, but here is the fix: https://github.com/antonblanchard/microwatt/pull/399

mikey, 2022-08-29:
closed with #399

closed 2022-08-29 by mikey