antonblanchard/microwatt #431: OpenPOWER spec covers both big-endian and little-endian

yskelg, 2024-08-18:
I propose adding support for big endian systems to the microwatt project. This would significantly expand the reach of the project and allow more users to benefit from its features.

Currently, the microwatt project only supports little endian systems.

OpenPOWER spec covers both big-endian and little-endian.
https://openpowerfoundation.org/specifications/64bitelfabi/

**Justification**

* **Increased user base:** Supporting big endian systems would attract a larger user base, especially those working on embedded systems with specific hardware(focused on network processing) requirements.
* **Platform independence:** By supporting both little endian and big endian systems, microwatt would become a more platform-independent solution.

**Possible approaches:**

* **Conditional compilation:** Use preprocessor directives(macro) to conditionally compile HDL code based on `Makefile` argument.

I would be happy to contribute to the project by providing code or documentation to support this feature.

Thank you for considering this request.

paulusmack, 2024-08-19:
Microwatt already has full support for both big-endian and little-endian execution modes, selected by bit 63 (the least significant bit) of the MSR, as defined in the Power ISA. This support is exercised in the 'modes' test. You could look at the source code in tests/modes if you want to see examples of how to switch between little-endian and big-endian modes.

Do you have any specific software in mind that you would like to run that needs to run in big-endian mode?

yskelg, 2024-08-20:
Thank you so much Paul,

Oh, I was following the `README.md` on the Arty A7 board and doing the lab when I mistakenly thought it only supported little-endian Linux.

I haven’t tested the big-endian Linux kernel and applications on my FPGA board yet.

Thanks for letting me know. I'm planning to use big-endian Linux according to the OpenPOWER architecture specs.

The goal is to test which performs better in network throughput, little-endian or big-endian.

I added a strikethrough on the issue where I made a mistake, just in case anyone else gets confused and thinks it's not supported. If everything works fine on big-endian linux, I'll update the #430 document as well.

I appreciate it. I'll do some research and testing based on what @paulusmack guide me. 👍